sinara-hw/mirny #1: Mirny schematics review

CannaCardo, 2018-07-25:
I sketched initial version of schematics ([link](https://github.com/sinara-hw/mirny/blob/master/PCB/Mirny.PDF)).
I've chosen [ADF4351](http://www.analog.com/media/en/technical-documentation/data-sheets/ADF4351.pdf)  ([HMC833](http://www.analog.com/media/en/technical-documentation/data-sheets/hmc833.pdf) has higher price and twice the power consumtion).
I left EEM connectors since I'm not sure which alternative (PoE/ I2C from Kasli and 12V from somewhere/ something else) would be better. 
Also I've left out frequency multipliers for now. I understand that this section should be realized as separate microwave PCB and it can be designed separately. I couldn't find any off-the-shelf 40MHz-4GHz multipliers, so that would mean designing it from the beginning or perhaps using a lower bandwidth doubler/tripler?
I appreciate any suggestions and having my mistakes corrected.

gkasprow, 2018-07-25:
Do we really need CPLD here? In case of Urukul it simplified a lot, but here we have only simple SPI. LVC138 should do the job.
Common reference is also good idea. I'm pretty sure that passive divider would do the job. Check what is required power level at the REFin. 
Do we need external sync input? We could simply copy clock distribution from Urukul.
If we apply divider at the VCO output, we will end up with plenty of harmonics since the output is just a square wave. Maybe we should let the user to apply the lowpass filter. It can be attached to the SMA directly. Mini-circuits offers broad selection of them.
@jordens going above a few GHz means that we have to use Rogers material which is quite expensive. We can mitigate it by using castellated RF modules dedicated to certain band. I'm a bit sceptical about  amplifier/attenuator working in 35MHz-13GHz range.

gkasprow, 2018-07-25:
Edit. It's not that bad with wideband attenuators and amplifiers.
Let's have a look at:  [HMC424ALH5](http://www.analog.com/media/en/technical-documentation/data-sheets/HMC424ALH5.pdf),  [ADRF5720](http://www.analog.com/media/en/technical-documentation/data-sheets/adrf5720.pdf), there are also plenty of amplifiers that work from almost DC to >20GHz, both from ADI and MiniCircuits. The most problematic components are baluns, but we can use asymmetrical output from the VCO chip. The VCO chip has also programmable output power but from only -4 to 5dBm, I'm not sure if this is sufficient.

gkasprow, 2018-07-25:
I think that EEM is the most capable solution since it gives the user a choice. Once the  Humpback is finished it would give the user freedom in interface selection - USB, Ethernet or DIO.

jordens, 2018-07-26:
@gkasprow 
* I'd also prefer to get by without the CPLD. But there are the 4 RF switches, the attenuators, the clock mux, the output status flag from the synthesizers, the LEDs.
* We don't need the sync circuitry. Just clock.
* Right. The harmonics filters shouldn't be fixed or populated by default.
* I wouldn't bother with the ranges above 6 GHz on the basic Mirny either. Leave footprints for MMCX connectors to expose that to custom boards. Maybe we can make Miny smaller in length or width so that those boards can be next to Mirny and don't need to be mezzanines. Consequently I wouldn't bother with the ultra wideband amplifiers and attenuators. Doesn't proper matching get extremely hard if you want to do that from 30 MHz all the way up to 20 GHz?
* I'll also vote for EEM connectivity.

gkasprow, 2018-07-26:
- CPLD seem to be simpler than bunch of serial registers and gives greater flexibility. Let's leave it then.
- sure, we don't need sync.
- what about addding MMCX just about the VCO chip for one who wants to play with their own AFE. We can use VCO's second output for this purpose.
- let's expose some CPLD pins and supply for control of such AFE board. The board could sit on top of Mirny and expose its connectors via the front panel. Maybe 4HP would be enough.
- on board we install just amp, attenuator and switch for let's say 6GHz where FR4 still does the job. But the VCO can work much higher.
- you are right, playing above a few GHz is quite challenging but there are great tools to do so. Have a look at https://www.xmicrowave.com/products/

hartytp, 2018-07-30:
> I think we should survey the users and use cases to decide about:

The use-cases I can imagine for this in our group are.
1. Driving microwave EOMs up to 3.2GHz (calcium hyperfine splitting). This requires a tuneable, but not frequency agile, microwave source with amplitude control and (real-time) switching.
2. Providing a fixed-frequency LO for upconversion. Similar requirements to above

So, my answers to @jordens questions would be:
  - Since we'd want some way of controlling the RF switches from Kasli in real time, EEM seems the right solution for us, but LVDS-over-RJ45 + external power (PoE?) could also work. We could live with ethernet only, and add an external switch, but that's less convenient. I hate USB for lab stuff
  - My preference for the reference would be 125MHz from Kasli via an internal MMCX cable like Urukul. We could probably get away with an XO, but it's nice to have it referenced to our atomic reference so that one doesn't need to check the frequency ever. Since we will generally use this in a rack with Kasli, it seems silly not to use the reference outputs that Kasli provides
  - Power: either EEM or PoE, depending on what we decide above seems easiest
  - We don't need much in the way of fancy features

jordens, 2018-08-01:
* Let's do EEM (will need to be dual slot again for high resolution RF switch signals) and push the rest (Ethernet, PoE, etc down to Humpback/Kasli)
* You'll be checking the frequency anyway due to the finite resolution of the PLL, even with a perfect reference the output won't be what you request in many cases.
* I'd like to leave the XO on board for Humpback use cases or where the MMCX cabling is unnecessary and annoying. And I'd like to make the XO the default. I think it would be best if on one input of the mux there is the SMA and the MMCX in parallel with a capacitor population choice (or even just hve them in parallel, the stub is probably small enough to not matter). And on the other input the XO. That covers all use cases that were mentioned.
* Maybe we can find a less power hungry and cheaper fanout (we don't need the 8 outputs or the divider features).
* Yes. Expose the high frequency (multiplied) output of the VCO on MMCX (DNP FWIW) for custom AFE purposes. Keep the base board good to 4 or 6 GHz.

hartytp, 2018-08-01:
> Let's do EEM (will need to be dual slot again for high resolution RF switch signals) and push the rest (Ethernet, PoE, etc down to Humpback/Kasli)

:+1: 

I could definitely live without the high resolution switching/second EEM for this design. But, given that we have the CPLD, if we're going to have one EEM connector, the overhead in adding a second is so low that there doesn't seem any point not having it.

hartytp, 2018-08-01:
> We might want to put a bit more power into the preamplifiers (20dBm output instead of 10dBm) since Mirny is not ad hungry as Urukul.

For my money, 20dBm might be a bit on the high side. One has to assume that sooner or later (and probably sooner) the max output power will be applied to the load due to software bugs. 20dBm is starting to get to the point where it can damage things.

In fact, 20dBm falls in to an awkward gap for us that makes it nasty: it's not enough power to drive our resonant microwave EOMs (given the typical modulation depths we need that needs >0.5W and typically more like 1W). Thus, we'd still need PAs for those cases. But it is plenty of power to trash a PA such as ZHL-03-5WF+. So, you'd end up having to stick a 10dB attenuator after Mirny.

For anything like an IQ mixer, 20dBm is way too much.

So, my question is what's likely to be a more common situation: 20dBm being enough power to avoid the need for a separate amplifier after Mirny, or 20dBm being too much power for whatever load is present, requiring the user to add fixed attenuators between MIrny and its load. e.g. we currently have stacks of MCL attenuators attached to Urukuls to avoid the possibility of damaging amps/AOMs -- this is one of my key motivations for Booster.

hartytp, 2018-08-01:
> then there are some really nice 4-input LVDS muxes e.g. from Si Labs. M

At least, I thought there were, but I'm struggling to find them now. @gkasprow I'm sure we found some in the past, can you remember where from?

Otherwise, I'd probably still prefer to cascade 2 clock muxes to make a 3 input mux to ensure that all options are supported without hardware rework. We can use relatively cheap, low-power LVDS ones.

gkasprow, 2018-08-01:
Maybe it was [this](https://docs-apac.rs-online.com/webdocs/1327/0900766b8132728c.pdf) or [this](https://www.onsemi.com/pub/Collateral/NB6L572M-D.PDF) ?

gkasprow, 2018-08-01:
I also have [this one](http://ww1.microchip.com/downloads/en/DeviceDoc/sy89544u.pdf) in my libraries

hartytp, 2018-08-02:
@gkasprow thanks! Any of those should work out well. My thoughts are:
- AFAICT, unlike the AD9910, the ADF4351 does not need a reference divider to work with 125MHz input clocks, as can take input frequencies up to 250MHz directly
- the ADF4351 has a single-ended input. So, assuming we're not planning to put a balun between the fanout buffer and the input, we will drive it from just one half of a differential output stage. So, to drive 4 synths, we only need a 2-output clock buffer. So, let's rule out the SY89544U, as it only has a single output so we'd need a fanout buffer after it. From the ICs you posted above, that leaves the NB6L572M or the SY89547L
- The ON chip is better specified, but more expensive and less available. So, I'd probably go for the SY89547L.
- looking at the ON chip phase noise, and comparing to figure 9 of the ADF data sheet with divider 32 (i.e. roughly 125MHz synth output, NB this is 63kHz loop BW for the ADF). We see:

offset |  ON | ADF
--- |  --- | ---
10kHz | -134dBm/Hz | -125dBm/Hz
100kHz | -136dBm/Hz | -130dBm/Hz
1M | -149dBm/Hz | -160dBm/Hz

- So it looks like the ON chip won't be a limitation. I'd guess that's also true for the Microchip one (or any other fast LVDS buffer for that matter) as the jitter specification is basically the same as the ON chip.

hartytp, 2018-08-02:
NB the microchip IC requires 3V3. If that's an issue we should use the ON chip and not bother with an extra LDO. AFAICT there are no microchip 4:1 muxes with two outputs and I don't want to have to add a fanout buffer to this (we *might* get away with using a passive splitter network to drive 4 synths from a single clock buffer output, but the signal level could get a bit marginal).

hartytp, 2018-08-16:
out of curiosity, what's the timeline for Mirny? Is this something that's likely to be completed soon?

hartytp, 2018-08-17:
@gkasprow thanks for the info. Sounds like a new toy for next year :)

CannaCardo, 2018-12-02:
I've added SY89546U as SY89546U output swing was not high enough for ADF4351 ([schematics](https://github.com/sinara-hw/mirny/blob/master/PCB/Project%20Outputs%20for%20Mirny/Mirny.PDF)). As for the front panel looks like 4hp is enough [(PDF3D)](https://github.com/sinara-hw/mirny/blob/master/PCB/Project%20Outputs%20for%20Mirny/Mirny_PDF3D.PDF).
The distance between mezzanines and pcb is 9mm (the same connectors as in SAYMA AFEs). Lightpipes used are: [this](https://pl.mouser.com/ProductDetail/Bivar/PLP1-100-D?qs=bY1dm4K8r4UTaxBhWQ5txQ%3D%3D) and [this](https://pl.mouser.com/ProductDetail/Bivar/RLP1-400-800?qs=%2fha2pyFaduhAd%252brRE3O7RxdFiKFLFnN2QTWOQtmi4m3%252bBxLAqcn%2fbQ%3d%3d). The side emitting SMD LED with lightpipe could be replaced with single bent THT LED but i'm not sure what would be better here. 

Assumming AFE mezzanine setup like [HMC424ALH5](https://www.analog.com/media/en/technical-documentation/data-sheets/HMC424ALH5.pdf), [ADRF5720](https://www.analog.com/media/en/technical-documentation/data-sheets/adrf5720.pdf) and [EHC-24L+](https://ww2.minicircuits.com/pdfs/EHC-24L+.pdf) we could get away with 3 supply rails: +-5V and +3.3V for digital, or +-7.5 leaving some headroom for LDOs.
The current consumtion would be around +20mA, -10mA + control circuitry per board. With current setup we can supply +7.5V@34mA, -7.5V@60mA per board. This is not yet included in power budget on schematic.
As for IO i've copied [#8](https://github.com/sinara-hw/sinara/issues/8) with only difference being 14 instead of 16 lines as i've run out of CPLD pins and it should be plenty anyway.

gkasprow, 2018-12-02:
- [ ] the shielding clips may be kept not mounted. We will probably remove them after tests. However they may be needed when working with mezzanines.

gkasprow, 2018-12-02:
- [ ] 3V6 rail is filtered rail. Please don't use it to supply any power modules. Is there any reason why LTM8045 cannot work from 12V directly?
- [ ] half of LTM4622 is wasted. Maybe it could make sense to use it to deliver 5.5V that then is used to create 5V by IC20? We would limit losses without changing BOM.
- [ ] please make sure all onboard LEDs have similar current. Big changes in LED current make impression that one of the power rails is broken.
- [ ] Mezzanine power consumption is not included in BOM. Lots of RF circuits work with 5V. So we can use second half of LTM4622 do deliver higher current 5.5V to mezzanines together with -5.5V or -6V from LTM8045. So the voltages could be: 3.3, 5.5, 7.5(maybe), -5.5. Any objections? LTM4622 do not support 6V.
- [ ] the ring around the M2.5 hole is too small. Typical stand-offs require at least 5.5mm
- [ ] SMPs require 9mm board spacing. The same with connectors that were used in Sayma RTM. Please confirm.
- [ ] you should not run high speed traces in such way:
[image]
Run both of them as 50R ones. Now you run them as coupled pair with let's say 2x70Ohm that forms 100R diff and then without changing thickness you route them as SE still as 70Ohm trace.
- [ ] In this way you will get extreme current density. I will show you how to run PI test
[image]

hartytp, 2018-12-03:
Good work @CannaCardo! I'll do a design review soon and provide you with some feedback.

CannaCardo, 2018-12-03:
Added mounting hole close to SMA, changed pad diameter to 5.5
[image]
[image]

I2C tree added to the schematic, for now i've assumed there'll be an EEprom chip for identification (temperature sensor probably isn't necessary as they will consume little to no power).

VCO PLL loop was designed in ADIsimPLL, parameters are now on schematic.

> AFAIR the ADF4351 has faster, pin-compatible versions. Can you please make sure that they are 100% pin compatible and if not, add necessary components to make them. Do it if it is just one or two components. If it is more complex, leave it as it is, but document what need to be changed.

ADF4355/ADF4356/ADF5355/ADF5356 are pin compatible, but not with ADF4351/ADF4350. They all require 5V supply apart from 3.3V, RFoutA is shifted by 1 pin and few of pins are also switched from supply to ground or to a decoupling cap. Their input sensivity is lower (400mVpp).

> Why SY8955 has 100R pulldowns? in LVPECL usual value is 200R.

The pulldown values for SY8955 are from the [datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/sy89855u.pdf) (page 11). 

> in some scenarios, only first EEM will be used. So do not relay on second I2C interface.

fixed

> the shielding clips may be kept not mounted. We will probably remove them after tests. However they may be needed when working with mezzanines.

Changed shielding clips to DNP in base variant

> 3V6 rail is filtered rail. Please don't use it to supply any power modules. Is there any reason why LTM8045 cannot work from 12V directly?
half of LTM4622 is wasted. Maybe it could make sense to use it to deliver 5.5V that then is used to create 5V by IC20? We would limit losses without changing BOM.

LTM8045 has highier efficiency at lower voltage input (with lower max output current which is not a problem here). At 12V it'll drop by 10-20% depending on output current. At 3.3-5V it's pretty consistent.
5.5V from LTM4622  Sounds like a good idea, could be also used by LTM8045.

> please make sure all onboard LEDs have similar current. Big changes in LED current make impression that one of the power rails is broken.

Changed resistor value for LD11, should be good now.

> SMPs require 9mm board spacing. The same with connectors that were used in Sayma RTM. Please confirm.

Yes, board spacing is 9mm with SMPs and LSS-120-01-L-DV-A, LSS-120-02-L-DV-A connectors.

gkasprow, 2018-12-03:
When thinking about power efficiency, take into account combined power efficiency. So if you multiply PE of first stage by PE of second stage, the combined PE would be similar.

jordens, 2018-12-04:
Thanks @CannaCardo !

* Could you add classic 2d board renderings, renderings of the gerber layers, stackup info like @gkasprow does in the PDFs he generates? Those 3D PDFs are not usable to some of us.
* Can we go for thinner lightpipes that are more similar to the DIO_SMA/BNC, Sampler, Zotino LEDs (not the big Urukul LEDs)?
* The I2C switch address should be 1110010 (see Banker, don't collide with the switches on Kasli)
* Do TR2/TR3 work all the way down to 10 MHz?
* The phase flipping of two channels is fine AFAICT the output phase won't be synchronized anyway and the phase relation between reference/PFD spurs of different channels doesn't matter to me.
* @gkasprow I would do 3 individual and 2 common signals to the AFEs (instead of 14 common).
* I would assume a reference frequency of 100 or 125 MHz for the loop filter. Can we make OSC1 a 100 MHz?
* Is DC coupling on CLKIN1 really OK?
* I would remove and bypass FL1 (like you had before). For most output frequencies this filter won't filter harmonics. And they are not an issue in most cases anyway.
* @gkasprow We needed it for Urukul to get precision timing on the RF switches and deal with the quad SPI mode. But here I think we may want to try to get rid of the second EEM. Let's use the CPLD in a "prefix" mode where you select the desired SPI bus (attenuators, configuration, clock select, LEDs, status, RF switch register, PLL, AFE ...) with the first 8 bits of the SPI transfer and then the actual transfer follows. I think that can be done in the CPLD. Then the only EEM connector would be wired as: CLK, MOSI, MISO, CS, SW0, SW1, SW2, SW3. You can set that in stone (no direction toggles and all pins except MISO are LVDS inputs). That frees up tons of pins (24) on the CPLD that can now go to the AFEs.
* You are right, the ADF435[10] and ADF535[56]/ADF435[56] are slightly different. Enough to choose one family. I am a bit unsure wheter it is worth to go to the high, output frequency ranges of the more expensive family. The Rubidium and Caesium people would like it. If we stick with the AD435[10] then there is little need for the AFE and the RFoutB stuff since the RFoutA covers the full range. I would gravitate towards the ADF535[56]/ADF435[56]  family.
* I'll leave it to @hartytp and @gkasprow to select a good clock mux/fanout here. Just make sure it doesn't degrade the PLL noise significantly.

CannaCardo, 2018-12-06:
I don't think fixed direction on EEM is necessary, as just by removing the other EEM we can do 4  individual and 4 common signals to the AFEs with 10 pins left (i guess we could go for 6-6).

> When thinking about power efficiency, take into account combined power efficiency. So if you multiply PE of first stage by PE of second stage, the combined PE would be similar.

Ack, LTM8045 is now supplied from 12V, output voltage changed to -5.5V. Second half of LTM4622 outputs +5.5V to AFE boards and IC20.

> Could you add classic 2d board renderings, renderings of the gerber layers, stackup info like @gkasprow does in the PDFs he generates? Those 3D PDFs are not usable to some of us.

Done

> Can we go for thinner lightpipes that are more similar to the DIO_SMA/BNC, Sampler, Zotino LEDs (not the big Urukul LEDs)?

We can replace short lightpipes with [1mm](https://pl.mouser.com/ProductDetail/Bivar/PLPC1-188?qs=sGAEpiMZZMuu2qEGwSNRoCaOwCO77pB5Q%2fFJeKNts5ZXXErwVZfpWg%3d%3d) or [2mm](https://pl.mouser.com/ProductDetail/Bivar/PLPC2-188?qs=sGAEpiMZZMsnnjF5aXV%2fXhA%2ftizPYRtVvS1h18e%252bW48%3d) ones, but the right angle one is the smallest diameter i could find that would go over SMA. Imo if we don't replace all of them then one larger LED will look odd.

> The I2C switch address should be 1110010 (see Banker, don't collide with the switches on Kasli)

I2C switch address changed to 1110010 (0x72).

> Do TR2/TR3 work all the way down to 10 MHz?

Yes, TCM2-43X+ frequency range is 10MHz - 4GHz

> I would assume a reference frequency of 100 or 125 MHz for the loop filter. Can we make OSC1 a 100 MHz?

We could replace it with something like this [100MHz, 5ppm](https://pl.mouser.com/ProductDetail/SiTime/SIT5021AI-2DD-33E-100000000X?qs=sGAEpiMZZMsBj6bBr9Q9aYNmGJvSVUJbwzsfzEAyMgTL%2f4Xdt6p%252bWw%3d%3d) oscillator.

> Is DC coupling on CLKIN1 really OK?

SY89855 can also work with DC coupled inputs, in that case VT is left floating.

> I would remove and bypass FL1 (like you had before). For most output frequencies this filter won't filter harmonics. And they are not an issue in most cases anyway.

Done

jordens, 2018-12-07:
* As long as there are free pins on the CPLD, I'd increase the number of individual IOs and also convert common IO to individual IO.
* That 100 MHz osc looks good. Also needs loop filter adaption.
* I suspect that the lowest SMA (clock in) will interfere with the front panel handle. This may require shuffling all the LEDs and SMAs again.
* And I still would gravitate towards the ADF535[56]/ADF435[56] family. Otherwise the AFE stuff isn't really necessary since that auxiliary output isn'd much different from the primary one.
* Ack the rest.

jordens, 2018-12-07:
* Hmm. I couldn't find any info on the 100 MHz OSC noise. We need to select that and the MUX to not degrade the PLL/VCO noise too much.
* I would make space and add the footprint for the cPCI S.0 backplane connector. @gkasprow ?
* If the four individual AFE becomes too small and tight, especially with the front panel SMAs, just double them up into two pairs, each with a single mezzanine connector. I also think that bulkhead connectors with pigtails would be fine here if that helps with mechanical layout. I would also leave the panel tutouts for the AFE outputs filled. Since there is no AFE yet, that allows the first AFE user to just drill them and put their AFE in.
* I also suspect that there will be similar mechanical overconstraint issues as for Sayma. THere is the AMC gasket edge, and the horizontal constraint due to the frojnt panel as well as the vertical constraint of the SMP/mezzanine connector. Maybe doing pigtail would really help here. 
* AFAICT, Master_RESET serves no function and can go.
* If possible please also route the 12V rail to the mezzanine connector. I expect people want to add medium power amplifiers to the mezzanine.
*

CannaCardo, 2018-12-16:
> As long as there are free pins on the CPLD, I'd increase the number of individual IOs and also convert common IO to individual IO.

Ack, with 2 common and 8 individual signals to AFEs all free CPLD pins are used.

> Hmm. I couldn't find any info on the 100 MHz OSC noise. We need to select that and the MUX to not degrade the PLL/VCO noise too much.

I've replaced the oscillator with [this](https://www.digikey.pl/product-detail/en/ecs-inc/ETXO-H33CL-100.000/XC2134-ND/5875502) one and I've copied Urukul clocking scheme. 

> I suspect that the lowest SMA (clock in) will interfere with the front panel handle. This may require shuffling all the LEDs and SMAs again.

I've pushed everything away from the handle, currently there is 1.2mm clearance between top of the handle and SMA cable nut for the lowest SMA.

> And I still would gravitate towards the ADF535[56]/ADF435[56] family. Otherwise the AFE stuff isn't really necessary since that auxiliary output isn'd much different from the primary one.

IC9 changed to ADF4355. Its the cheapest one and if needed can be swapped without any other changes.

>  I would also leave the panel tutouts for the AFE outputs filled. Since there is no AFE yet, that allows the first AFE user to just drill them and put their AFE in.

I've added panel file without the cutouts for AFE boards.

> If possible please also route the 12V rail to the mezzanine connector. I expect people want to add medium power amplifiers to the mezzanine.

Done

jordens, 2019-01-03:
* Check the output bias inductors. https://freenode.irclog.whitequark.org/m-labs/2019-01-03#1546514242-1546514169; Optimize for wideband match.

hartytp, 2019-01-12:
Why did you decide to go for the adf4355 in the end rather than the adf4356?

hartytp, 2019-01-13:
@CannaCardo some relatively minor comments:
- [ ] it's convention to have traces to ground going down. e.g. having looked at a lot of schematics, I find it mildly jarring that the output capacitors around IC16 go upwards from the trace (same for the LED)
- [ ] in the power budget, can you add a column heading to make it clear which numbers are mA and which are mW etc
- [ ] before going to manufacture we need to finish debugging the power supply issues on urukul related to the LT1763
- is it really necessary to have 4 separate mezzanines rather than just having a single 4-channel mezzanine? IME having lots of narrow mezzanines is mechanically worse and makes layouts nasty on the mezzanines (lots of dead space around boards, can't share common components between boards). What do you think about our idea for stabilizer of having the mezzanine have its own FP and using pin header for the board-board connections (as well as the SMPs)?
- why are there NC pins on the mezzanine connector? Better to ground them?
- [ ] can we reserve some pins on the FPGA for HW rev?
- [ ] PN for TR1? What frequency range does it cover?
- [ ] worth picking a filter that goes up to the 6.8GHz fundamental VCO range? (or at least to the max BW for other components like switches etc)
- [ ] looking at the eval board schematic for the ADF5356, I can see a few differences. e.g. capacitance vales (places where they've used, say, a 10uF and you've used 0.1uF for decoupling). Also, some power pins they've run from separate LDOs, where you've run them from the same LDO. Please double check the eval schematic and make sure that all differences are for a good reason

sbourdeauducq, 2019-01-13:
Probably too late for this design, but in the future let's try to use Lattice FPGAs instead of Xilinx CPLDs (AFAICT they would do the job there), so we can get rid of ISE/Vivado.

jordens, 2019-01-15:
Stick with the CPLD.
IME the free tools for this job are not ready yet. There are still lots of problems and they don't make me nearly as productive as ISE for CPLDs. Also from what I have seen, at least for this application, the CPLD is faster and has more reproducible delays than the ice40.
Vivado won't go. We will be needing Vivado for the foreseeable future.
I'd rather spend my time fixing design flaws and improving the project than trying to get rid of something that we definitely won't get rid anytime soon.

hartytp, 2019-01-15:
One other question: do we really want to bother with all the complexity/mechanical issues around RF mezzanines for this project? Can we get away with something like using solder jumpers to swap between two paths: RFA + AFE; and RFB directly at the output?

If we do need AFE mezzanines, another option that might be simpler/better would be to use a similar form-factor to Stabilizer AFEs (separate EEM FP, pin-header for board-board connectors) and use short coax leads to internally route the microwave output from Mirny to the AFE. This could be simpler and more robust than trying to make the stackup work with SMP bullets.

jordens, 2019-01-15:
I agree. That's what I argued for in https://github.com/sinara-hw/mirny/issues/1#issuecomment-445353727 And a single or at most two mezzanines is fine IMO.
But I would not try to capacitor-route RFoutB to the front panel. Just MCX/MMCX and a pigtail to the panel would probably be electrically better and simpler. For the AFE I also don't like squeezing it into the 4HP space and I don't like the constraint mechanics and interference due to panel/rf/digital connectors.

hartytp, 2019-01-15:
Thakns @jordens. So in order of preference it would be (a) scrap the AFE and use a coax pig-tail (connectors chosen to choose the frequency range) to route RFB somewhere accessible (b) if we have to keep the AFEs then let's limit it to just 1 AFE.

For the pig-tail approach, I assume you'd route them to an adjacent 4HP FP that, for example, just has 4 feedthrough SMAs mounted?

CannaCardo, 2019-01-15:
> can we reserve some pins on the FPGA for HW rev?

I've added 2, should be more than enough.

> PN for TR1? What frequency range does it cover?

TCM2-43X+,  it covers 10MHz-4GHz. Added PN on schematic.

> worth picking a filter that goes up to the 6.8GHz fundamental VCO range? (or at least to the max BW for other components like switches etc)

LFCN-3800+/ LFCN-4400+ would cover BW of front end components (everything works up to 4GHz).

> looking at the eval board schematic for the ADF5356, I can see a few differences. e.g. capacitance vales (places where they've used, say, a 10uF and you've used 0.1uF for decoupling). Also, some [...]

Capacitance values and shared supply lines were not updated after the change from ADF4351. Should be good now.

hartytp, 2019-01-15:
> LFCN-3800+/ LFCN-4400+ would cover BW of front end components (everything works up to 4GHz).

What limits the frequency? How easy would it be to make the AFE work over the full VCO BW?

hartytp, 2019-01-15:
ack. Not worth a complete redesign of the AFE to get a bit more BW unless some user has an application that really needs that 4-6.8GHz band (but in that case they really need to speak up now).

jordens, 2019-01-15:
I think covering the entire range from <100 MHz up to 6.8 GHz reasonably well will be very difficult. I'd rather have people either use the RFoutB AFE for > 4GHz or modify the board themselves.

hartytp, 2019-01-15:
ack

hartytp, 2019-01-15:
@gkasprow I agree. Expose the RF and DIO on sensible connectors and add some mounting holes.

CannaCardo, 2019-01-15:
Should the IDC be made incompatible with EEM connector to avoid possible confusion? Or would clear silkscreen be enough to shave additional BOM line?

hartytp, 2019-01-15:
> Should the IDC be made incompatible with EEM connector to avoid possible confusion? Or would clear silkscreen be enough to shave additional BOM line?

Yes! cf what we did on stabilizer.

I'd try to position the IDC so that it can either be used as a board-board header for a long mezzanine card (a la Stabilizer) or used with a ribbon cable for a shorter mezzanine (as Greg suggested above). Basically, kick this decision into the long grass.

jordens, 2019-01-15:
I'd still put that mezzanine IDC to the right of the PLLs, out of the RF zone, either near the top or near the bottom to not interfere too much with the center area. Could also be horizontal instead of vertical.
Also not to far to the right to not interfere with the cPCI S.0 adapter.

Keep the board at 160 mm for now. Move the EEM IDC connector to the designated place (for the cPCI S.0 adapter board, see the other recent designs of @gkasprow ) If (a) it is really easy and (b) does not interfere with the EEM connector position, (but only if both are met) add the cPCI S.0 connector but DNP it.

gkasprow, 2019-01-15:
simply copy position of the EEM connector from Stabilizer. Take into account that CPCIS adapter will have two connectors for double EEM boards so make some space as I did on the Stabilizer board.

CannaCardo, 2019-01-16:
IDC used in Stabilizer is 16pin, currently we have 28 DIO lines, I2C and 5 voltage rails (-5.5, +3.3, +5.5, +7.5, +12) supplied to mezzanines. Assuming we keep all rails and take 2 pins for ground (which imo would be on the short side) that leaves 9 pins for DIO and I2C. We could either
- leave it as it is if that's ok,
- drop some power rails/DIO for ground (I'd say 7 DIO lines would be enough for SPI+driving 4 switches or something similar, traces could be matched in length),
- add 2nd IDC connector/ pick larger one.

cPCI S.0 connector would interfere with bottom mounting hole for adapter and EEM connector. 
[image]

gkasprow, 2019-01-16:
we will use 220mm CPCIS boards anyway, just use holes and IDC connectors placement as in Stabilizer.

CannaCardo, 2019-01-19:
MCX/MMCX connectors work up to 6GHz ([see](https://www.amphenolrf.com/frequency-range-chart)), if we want to pass up to 12.8 GHz to AFE board we need to pick something different, I've left SMPs for now.  

- EEM connector moved and mounting holes added for CPCIS adapter,
- AFE connector changed to IDC,
- SMAs changed to angled ones,
- lightpipes dropped since they're no longer needed,
- 0R resistors added to tie SMAs and panel to ground like in Urukul,

CannaCardo, 2019-01-22:
I've rerun the [DC drop analysis](https://github.com/sinara-hw/mirny/files/2781751/report.pdf) and did the thermal simulation.
The environment setup:
[image]
Results:
[image]
Same conditions but flow rate at 130cm/s:
[image]
Thermal cutoff would be at about 15cm/s flow rate (IC12 at 80deg):
[image]

hartytp, 2019-01-22:
@CannaCardo can you confirm which PLL IC you're planning to use? Is it the ADF5356?

Also, we should consider adding the pads for a (DNPd) active (3rd order) loop filter.

jordens, 2019-01-23:
The users are specifically interested in frequencies around 6.8 GHz. They'll handle the AFE. Let's stick with the ADF535[56]. I don't think the price difference should decide that. Maintaining variants and the induced board shortages would more than make up for that.
Yes. just a bigger IDC is fine. Also, and again, I think the negative supply can really go away. That can be implemented on the AFE.

CannaCardo, 2019-01-23:
ADF535[56] RFoutB frequency range is 6.8 - 13.6 GHz (doubled VCO frequency) and RFoutA range is 53.125 MHz - 6.8 GHz (VCO divided by 1,2,4,8,16,32 or 64). With current AFE components there's a gap in 4 - 6.8GHz range.

gkasprow, 2019-01-23:
Guys, we have to converge quickly to the production version of the board. @CannaCardo has to finish writing his thesis and needs working HW and tests done. So let's leave minor tweaks for the second revision and proceed ASAP.

jordens, 2019-01-23:
The gap Is fine. People will need to tweak the rfouta path, or use the non doubled pll chips and a custom.afe.

jordens, 2019-01-23:
Imo there are only small things left (split attenuator spi, four layers, more Dio on mezzanine connector). See above. Some of them already mentioned a while ago.

CannaCardo, 2019-01-28:
Ok, I've updated schematic & pcb:
- changed to 4 layer stackup (updated [DC Drop Analysis](https://github.com/sinara-hw/mirny/files/2802634/report.pdf) )
- N5V5 supply removed
- AFE DIO increased to 8
- attenuators SPI is separated
- ADF4355 changed to ADF5356 as default
- added variants for other ADF chips

CannaCardo, 2019-01-28:
Done and done.

gkasprow, 2019-01-28:
@CannaCardo Please use OutJob file from Urukul project to generate the production documentation.

CannaCardo, 2019-01-28:
Ack, I've uploaded fabrication files

gkasprow, 2019-03-01:
for some reason the assembly house didn't know which chip to mount. I confirmed ADF5356

[1 more reply]

closed 2020-07-14 by CannaCardo