sinara-hw/mirny #2: power supply bug

gkasprow, 2019-01-12:
@jordens discovered [issue](https://github.com/sinara-hw/Urukul/issues/20) with the power supply in Urukul
The DC/DC converter has an output voltage higher than the specification says. We should look for another buck converter producing 7.5V
Another issue is bypass capacitor of LDO - with 100nF it oscillates.

gkasprow, 2019-01-28:
@CannaCardo 
- [ ] C170, C175 must be 100pF. Replace with the component from libraries, do not edit parameters with value.

CannaCardo, 2019-01-29:
fixed

hartytp, 2020-07-13:
@CannaCardo should we close? Which release is this fixed in?

CannaCardo, 2020-07-14:
It was fixed in v1.0rc4.

closed 2020-07-14 by CannaCardo

gkasprow, 2020-07-16:
@CannaCardo are you currently working on new revision or only are closing old issues?

CannaCardo, 2020-07-16:
Just closing, I'm working on the TDC atm.