sinara-hw/mirny #3: a few last minor issues

gkasprow, 2019-01-28:
- [ ] Please do not add DNP as the Mounted parameter because it is not transferred to the production files. Use variant instead and mark it as not mounted.
- [ ] remember to generate the production files with the appropriate variant enabled
- [ ] move JTAG connector above the ARTIQ logo. The existing location will be covered by the CPCIS adapter.
- [ ] add nice copper balancing as I did for the Banker. You will need this [file](https://www.dropbox.com/s/omaalewtysv5mm3/copper_balancing2.PcbDoc?dl=0). This is a simple hack. Lock all objects on Keep out layer. Then copy keep out mesh with polygons from the file I specified. The polygon parameters are critical. Regenerate polygons. Select the keep out layer only. Remove un-locked mesh from your design. In Xpedition, such copper balancing option costs 10k $ :D My hack is for free.

CannaCardo, 2019-01-29:
- All components marked as DNP are also marked as such in variants.
- There's a separate pick & place, bom and schematic file for every variant, gerbers and drill files are the same across all variants
- JTAG connector moved
- I've added balancing pours with a polygon cutout near rf sections, as floating copper there could cause some troubles later.

hartytp, 2020-07-13:
@CannaCardo this issue looks complete. Can we close?

closed 2020-07-14 by CannaCardo