sinara-hw/mirny #7: Mirny v1.0 tests

gkasprow, 2019-03-26:
The Mirny is ready for tests
[image]

gkasprow, 2019-03-26:
- [ ] 0R installed instead of  C68 ABCD causing short of 5V power supply rail. 3V3 rail depends on this voltage. After removing shorts, all supples are fine.

gkasprow, 2019-03-26:
CPLD is detected over JTAG

gkasprow, 2019-03-26:
@CannaCardo it's time to connect ADI devkit and check if the synthesizers are working

CannaCardo, 2019-03-26:
Ack, I should be in Warsaw on Thursday to test it.

jordens, 2019-03-26:
Nice!

gkasprow, 2019-04-04:
the CPLD cannot really set Hi-Z on CLK_SEL signal. the state is close to VCC.
With a touch of the probe, the Hi-Z appears and stays until power down. So it looks like additional init sequence would be needed that first sets low state and then Hi-Z.

gkasprow, 2019-04-04:
Other than that Mirny works with ADI devkit as a controller.

gkasprow, 2019-04-04:
We have some issue with 26kHz and its second and third harmonics. The carrier is 100MHz. Moreover, the analyzer sees the carrier shifted by 1kHz.
[image]

gkasprow, 2019-04-04:
It looks like the noise source was outside. With analyzer and supply connected to same mains outlet the spectrum looks much better now.
[image]

jordens, 2019-04-04:
It works on urukul. Just tristate it.

jordens, 2019-04-04:
What are the conditions and settings there? With div32 and a 3.4 GHz vco the datasheet gets to -131dbc/hz at 10khz and 64 mhz pfd.

gkasprow, 2019-04-04:
the SSA I have is not a really good one. It is a software option to the spectrum analyzer. It's good for quick debugging, not real measurements. 
With CMC added to the 12V power entry, the 26kHz peak disappeared.
We will make measurements with real SSA in a few days.

gkasprow, 2019-09-18:
[Here](https://www.dropbox.com/s/8773l9b8eklw5bs/praca%20mgr%20277384_inz.pdf?dl=0) is the thesis of Krzysztof Belewicz

jordens, 2019-09-18:
Well done. Congrats @CannaCardo

jbqubit, 2019-09-18:
Thanks for the detailed writeup @CannaCardo! Great work!

closed 2020-09-30 by hartytp