sinara-hw/mirny #10: [RFC] Phase lock application

hartytp, 2019-11-26:
We have an application where we'd like to generate two tones at 1.6GHz. One needs to be locked to a fixed external reference, the other needs to be phase locked.

It seems like Mirny might offer a nice way of doing this if we did something like:
- Add a footprint for a low phase noise VCXO which can be populated instead of the XO (or, better, aim for a pin-compatible part)
- Hook the tuning pin to an MMCX/whatever. Maybe add an RC filter or even a diff-amp to give noise rejection. All of this would be DNP by default
- have some means of getting some channels clocked from the VXCO and some from the external reference. This could be as simple as adding a DNP'd MMCX connector for each channel and solder jumpers that can be used to inject an external clock (that's somewhat useful for debugging anyway). 

Thoughts?

jordens, 2019-11-26:
To clarify: you want one voltage controlled Mirny channel and one controlled via a reference frequency?
Adding the tuning option to the XO is good.
And solder jumpers to select between DNP MMCX per channel and the clock fanout is also fine. But simpler yet and without any new connectors, you can use jumpers to choose between IC17:Q1-Q3 (VCXO or MMCX) and IC13:Q1-Q3 (#8, FP SMA or VCXO/MMCX) for three of the four channels. That way you can run 0 to 3 channels from the VCXO (or MMCX) and 4 to 1 channels from the SMA input.

hartytp, 2019-11-26:
> To clarify: you want one voltage controlled Mirny channel and one controlled via a reference frequency?

Yes. Basically, one as a reference for the mixer LO and one that we can lock.

We'll probably end up using Stabilizer for the lock so this could potentially be a Stabilizer mezzanine but the work to reward effort feels better by reporuposing Mirny.

> But simpler yet and without any new connectors, you can use jumpers to choose between IC17:Q1-Q3 (VCXO or MMCX) and IC13:Q1-Q3 (#8, FP SMA or VCXO/MMCX) for three of the four channels. That way you can run 0 to 3 channels from the VCXO (or MMCX) and 4 to 1 channels from the SMA input.

Yep, that's another route I thought of which I'd also be happy with.

hartytp, 2019-11-29:
Having said all that, this would be pretty trivial to do as a Stabilizer AFE card. 1 Crystal, 2 PLLs and some regulators and that's it AFAICT. Basically just a copy-paste from a few other designs so <1 day of work I'd guess.

gkasprow, 2019-11-29:
You can also connect Mirny to the stabilizer via EEM.

dhslichter, 2019-12-02:
Why not use Mirny as a carrier generator at ~1.25 GHz, split the output, and then use two separate mixers with two Urukul channels to mix up to 1.6 GHz?  The carrier and other sideband can be stripped easily with high-pass filters/notches, then all frequency and phase tuning between channels can be done digitally with the Urukuls.  If the Mirny can be locked to an external reference already, you're set.  You could make a little eurocard with splitter, mixers, filters, some SMA connectors on the front panel, even some ERA-4 or equivalent amps to boost the output power a bit if desired (without amps it could be a completely passive board, with amps you would have to provide a power supply from in the crate).

hartytp, 2019-12-02:
> Why not use Mirny as a carrier generator at ~1.25 GHz, split the output, and then use two separate mixers with two Urukul channels to mix up to 1.6 GHz?

(a) because there is a doubling cavity involved so we'd need 1.6GHz and a 3.2GHz tones (b) because it feels more expensive/complicated than hooking a stabilizer up to Mirny. What would do the digital modulation of Urukul? Kasli? I'd like this to be a standalone unit with good ethernet diagnostics (e.g. streaming noise traces). That feels easier to do with Stabilizer than Kasli.

I think the Vco tune input is a good thing to have for Mirny. But, currently I'm leaning towards doing this as a high-frequency version of Pounder with the DDSs replaced by ADF PLLs.

hartytp, 2019-12-31:
closing in favour of https://github.com/sinara-hw/Homodyner/issues/1 and https://github.com/sinara-hw/mirny/issues/11

closed 2019-12-31 by hartytp