sinara-hw/mirny #11: VCXO

hartytp, 2019-11-27:
The current XO is a crystek [CCHD-950](https://www.crystek.com/crystal/spec-sheets/clock/CCHD-950.pdf) costing ~£20 in singles

The a footprint-compatible VCXO part is [CVHD-950](https://www.crystek.com/crystal/spec-sheets/vcxo/CVHD-950.pdf). This is about the same cost (actually, very slightly cheaper) and has decent [availability](https://octopart.com/search?q=cvhd-950-100.000).

I'm happy to keep the XO as the default population option, but can we have either a variant with the VCXO or just an annotation mentioning that this is a suitable alternative part?

I'd like to break the tuning pin out to an MMCX. Probably via a difference amplifier (to remove ground small levels of common-mode noise), offset addition and attenuation to match +-10V to the VCXO's range. Then probably an RC filter to kill any high-frequency noise. I'll sketch out a suggestion for this design at some point when I have time.

hartytp, 2019-11-27:
@cjbe FYI this is the VCO noise spec https://octopart.com/search?q=cvhd-950-100.000

jordens, 2019-11-27:
I think we already went for the tunable part in Urukul and mid-pointed tuning input.

hartytp, 2019-11-27:
For reference, the PLL noise model is: 
- assume f_ref=125MHz, N = 13 (f_out=1.625GHz)
- in-band phase noise floor is `-227dBc/Hz +10log10(125e6) + 20log10(13)=-124dBc/Hz`
- FWIW, that's equivalent to reference noise of -146dBc/Hz
- 1/f noise is `-121dBc/Hz + 10log10(10kHz/f_offset) + 20log10(f_rf/1GHz)` so the 1/f phase noise at 10kHz offset is `-117dBc/Hz`

VCO noise is

[image]

hartytp, 2019-11-27:
If that's actually an issue and we ever made a mezzanine, we could consider using e.g. an HMC830 which has ~10dB lower noise, but I don't really think it's a problem. tl;dr it looks like this will work nicely!

hartytp, 2019-11-29:
FWIW, the HMC830 specs are 

[image]

So 3dB better for the in-band noise floor (although, note that the PFD can only hit 125MHz for integer-N mode)

I believe the flicker noise floor is normalized to 1Hz offset, so equivalent to 228 at 10kHz. i.e. approx 7dB better than the ADF PLL. So, not a huge difference between the two TBH. 

[image]

The ADF VCO noise actually looks a little better.

So tl;dr either part is probably fine.

gkasprow, 2020-09-30:
I copied the schematic from Urukul
[image]

closed 2020-09-30 by gkasprow