sinara-hw/mirny #13: transformer part numbers missing from schematic

hartytp, 2019-11-27:
Also, let's make sure these are done consistently with other designs like phaser.
- where possible/appropriate use same transformer part numbers as other designs
- for external clock inputs try to do the layout such that we can use the transformer to provide galvanic isolation (not an option for microwave inputs)

gkasprow, 2020-09-30:
what about the same transformers as in Phaser design?
[image]

hartytp, 2020-09-30:
fine by me

hartytp, 2020-09-30:
@gkasprow shouldn't the input ground be tied to the SMA ground, and only connected to the PCB ground through the 0R resistor though? That way we can use the transformers to provide galvanic isolation if desired (but let's not make that the default population option). IIRC that's what we're now typically doing elsewhere...

gkasprow, 2020-09-30:
ah, yes, good point.

closed 2020-09-30 by gkasprow