sinara-hw/mirny #19: Analog supply decoupling

hartytp, 2019-11-27:
Should there be FBs to isolate the various anlaog supplies that are derived from the same LDO?

hartytp, 2019-11-27:
Well, we're actually not far off the [eval board design](https://www.analog.com/media/en/technical-documentation/user-guides/EV-ADF5356SD1Z-UG-1087.pdf).

The only real difference I see is that the eval board uses separate LDOs for the output stages. I assume this isn't really worth doing and the FBs will be fine...

Other than that I didn't spot any issues during a quick design review. Nice job all who worked on this one!

gkasprow, 2020-09-30:
can we close it?

hartytp, 2020-09-30:
@gkasprow so long as you're happy with the current design (e.g. not using separate LDOs as the eval board does) you can close this issue. I'll leave the final call to you.

closed 2020-09-30 by gkasprow