sinara-hw/mirny #21: [RFC] OPLL AFE

airwoodix, 2019-12-16:
We are considering using Mirny to generate the ca. 7 GHz MW we use to mix down rubidium Raman phase lock beatnotes. Besides the necessary range extension (using RFoutB), it would be great to integrate most of the chain as a Mirny addon-board.

The full chain would be something like (x4):

```
PD input -- gain +16dB  -- DBM -- LPF (200 M) -- [PFD -- CP] -- LPF -- output
                            |
                     7 GHz from RFoutB
 ```

For the PFD/CP, something cheap and simple like ADF4154 or even ADF4110 would be great. The reference for the the PFD would come from an external Urukul channel. Since there're four lines free on the Mirny EEM, these can be routed to the AFE and control the PLL chip.

RFoutB probably needs preconditioning to saturate the mixer.

A simpler version of the board would drop the PLL if that's too complicated / not generic enough. In any case, a down-converted beatnote monitor is a useful feature.

What are your thoughts?

jordens, 2019-12-16:
A heterodyne design might be best. You would be able to reuse Pounder. 1st IF at ~100 MHz.

```
Mirny - gain - DBM LO
PD - gain - DBM RF - LPF - Pounder (gain, DDS LO, DBM, LPF, gain) - Stabilizer
```

hartytp, 2019-12-16:
@airwoodix Hi! BTW this is well-aligned with something we're thinking about.

We had a couple of thoughts about ways of doing this. e.g. https://github.com/sinara-hw/mirny/issues/10

As @jordens points out there are many ways of skinning a cat, but reusing Pounder seems like a good idea.

Currently, the option I'm leaning towards for our application would be to create a high-frequency (1-4GHz at least) version of Pounder. The idea is
- have the entire RF chain (filters, mixer, amps, variable attenuator etc) on a Stabilizer mezzanine. 
- replace the twin DDS on pounder with a pair of PLLs.
- one PLL uses an external reference
- the other uses a ~100MHz VCXO that's steered by the DAC on pounder to provide the phase lock

This provides a really nice integrated solution to the problem that can span a wide frequency range (can also have RF and LO at different frequencies if e.g. modulation is done before optical doubling). We can largely reuse the pounder design so the HW changes are relatively minimal.

jordens, 2019-12-16:
The other thing that I would consider in general for these phase measurements is going superhet and having some ~200 kHz IF (instead of DC) on Pounder and doing the rest in digital domain in Stabilizer. That's a free rejection of all the low frequency noise and instability near DC. Just needs proper clocking of Stabilizer.

hartytp, 2019-12-16:
@airwoodix what PFD/CP are you thinking of? Are you intending to lock the ADF to the beat note directly? Might be better to lock the ADF to a VCXO with a high bandwidth and then lock the VCXO to stabilize the beat note. The VCXO has much lower noise/better stability than the ADF VCO so you can use a narrower bandwidth lock, can do S/H etc

hartytp, 2019-12-16:
That's what we would do on our high-frequency Pounder version...

airwoodix, 2019-12-16:
Thanks for the feedback. Great ideas, I was looking at something much simpler (and cheaper).

From a pure accounting point of view, I'd priviledge a Mirny mezzanine since this is one board less (we need RFoutB gain anyway so the Mirny board is unavoidable).

@hartytp I was looking at ADF4154 as PFD/CP with no extra VCO but it sounds indeed convincing to add an extra VCXO. The main argument was to keep the system as simple as possible and the availability of lines on the EEM Mirny is attractive to me.

The phase lock's frequency offset needs to be dynamically controlled. The easiest would be over EEM, otherwise we'd need a lot of extra fw development for Stabilizer. Hooking up a Urukul channel or multiplexing a DDS next to the PLL chip on the Mirny addon board is easy and integrates seamlessly into the system (saving the indirection through Stabilizer).

@jordens of course, a superheterodyne approach is better but also more complex.

hartytp, 2019-12-16:
> Thanks for the feedback. Great ideas, I was looking at something much simpler (and cheaper).
From a pure accounting point of view, I'd priviledge a Mirny mezzanine since this is one board less (we need RFoutB gain anyway so the Mirny board is unavoidable).

FWIW my proposal is also only 1 board (stabilizer) + 1 AFE card so I don't think it will be more expensive than Mirny + AFE in terms of hardware production costs.

What are you planning to use to do the feedback controller? Will you do this in analog (boo!) or have a digital controller. I'd suggest you consider the latter. Not just because it's so much nicer to tune loop filters digitally (particularly if you might at some point want e.g. a double integrator to kill acoustic noise without having to have a really wide lock bandwidth etc), but because of the diagnostics. e.g. we've found it so useful to be able to stream error signals for diagnostics (see if a lock is ringing) do digital relock etc.

From that point of view using Stabilizer is the obvious choice for the lock (whether you use Stabilizer to drive the VCXO on Mirny or put the PLL on a Stabilizer AFE).

> @hartytp I was looking at ADF4154 as PFD/CP with no extra VCO but it sounds indeed convincing to add an extra VCXO.

I can't remember (don't have schematics in front of me) if Mirny even exposes the Vtune input right now to allow an external lock signal. So, locking like that might require hardware modifications.

> The main argument was to keep the system as simple as possible and the availability of lines on the EEM Mirny is attractive to me.

Stabilizer is an EEM too :)

Anyway, I take all your points. As I said, there are many ways of skinning a cat. I have my own ideas about how I'd do this but I'll be really interested to see what system you settle on. If you do a nice job of it then it would be a really welcome addition to the Sinara ecosystem.

airwoodix, 2019-12-16:
> FWIW my proposal is also only 1 board (stabilizer) + 1 AFE card so I don't think it will be more expensive than Mirny + AFE in terms of hardware production costs.
Given the frequency range, we need to use the ADF5356's second output (``RFoutB``). Would you expose it directly on the front panel from the SMP connectors? I'd take the opportunity to at least at gain (possibly with digital step attenuators driven by a SR like the PGIAs on Sampler).

We'd mostly use Stabilizers for the feedback (I totally support your points), but also likely Toptica DLC for the diode lasers (because we anyway use them for the current/PZT drivers). Hence the idea to put most of it on Mirny... Other experiments in the group might also be more convinced if they only need a Humpback and a Mirny + AFE (ok, with a lot of software, this is also valid for Stabilizer (as EEM upstream) + AFE + Mirny).

> I can't remember (don't have schematics in front of me) if Mirny even exposes the Vtune input right now to allow an external lock signal. So, locking like that might require hardware modifications.

It doesn't. Either we misunderstood ourselves or I'm stupid but I don't see the need to feedback on Mirny. I had an optical PLL "à la" Vescent D2-135 in mind so we'd feed back on the laser directly. Or is there something wrong with this approach?

>> The main argument was to keep the system as simple as possible and the availability of lines on the EEM Mirny is attractive to me.

> Stabilizer is an EEM too :)

Sorry should have written "on Mirny's EEM". With @jordens Mirny CPLD gateware, EEM4-7 are free to use on the mezzanine.

I'll work out the options tomorrow and come back with more questions :) Overall, I'd mostly be happy minimizing the amount of firmware/gateware to write. And honestly, as far as I'm concerned, it's easier to write glue gateware for the Mirny CPLD and a simple coredevice driver than extending the Stabilizer firmware.

hartytp, 2019-12-16:
> I had an optical PLL "à la" Vescent D2-135 in mind so we'd feed back on the laser directly. Or is there something wrong with this approach?

Aah, okay, makes sense.

I assumed you were using one output of Mirny to drive a modulator and wanted to do the optical phase lock by feeding back to the RF.

gkasprow, 2019-12-30:
Can we move it to the [new repo](https://github.com/sinara-hw/Pounder_HF) ?

hartytp, 2019-12-30:
@airwoodix can you sketch out a full block diagram of the system you want to build please?

 I have in mind a stabiliser homodyne afe based on pounder. Idea is to use one pll to provide a mixer ref. Lock is achieved either (eg population options) by using an external modulator which is driven by a second pll who’s ref is provided by a vcxo driven by stabiliser or using the stabiliser output to drive some other actuator (eg laser PST).

If there is enough overlap it would be good to design an afe that can cover both cases.

airwoodix, 2020-01-02:
I'd lean towards separating the frequency ranges (6G+ vs 3G-) and build two daughter boards: one for Mirny that deals with applications that require RFoutB and one like Pounder/Homodyner for the work in the 3G- domain.

For the 6G+ range, what about a Mirny AFE with chains like this:

[image]

The application is our typical phase lock for the Rb interferometry systems. The RF input is a beatnote at ca. 6.8 GHz. RFoutB covers the range 6.8 GHz - 13.6 GHz. Components suggestions try to cover at least up to the (neutral) Cs hyperfine frequency at 9.2 GHz.

RFoutB delivers 0-2 dBm, with -20 dBm fundamental VCO feedthough. The HPF doesn't cost much and probably makes this better. The mixer needs 17 dBm LO drive. All amplifiers have the same footprint. Would be great to keep the output LPF in a MCL FV1206 package to have more flexibility (and stay compatible with Urukul).

To achieve phase/frequency offset lock, the RF signal can be fed to Pounder as @jordens suggested. I think that my initial suggestion to put the phase detector on the Mirny AFE is not flexible enough although the full solution will require a significant amount of firmware development for Stabilizer.

I don't quite see the use for more than two heterodyning chains like above on one Mirny AFE so I'd be happier featuring the remaining two VCOs with standard conditioning chains (6G+ variant of what's on Urukul/Mirny baseboard):

[image]

The digital step attenuator is the tricky part. If one wants to cover above 7 GHz, the HMC424A is the only viable part I found this morning (anyone finds better alternatives?) and is annoying to drive (parallel, negative voltage control). Up to 7G, HMC305S would be suitable for example. HMC629A has only 3 dB LSB, which is IMO too coarse. The application for these outputs is to generate the base frequency for MW pulses. The output is tuned to max 10 dBm since most applications will require an extra power amp.

hartytp, 2020-01-02:
@airwoodix thanks for the detailed write up. That's interesting.

> To achieve phase/frequency offset lock, the RF signal can be fed to Pounder as @jordens suggested. I think that my initial suggestion to put the phase detector on the Mirny AFE is not flexible [...]

I see. So, the idea is to beat the IF from your Mirny AFE against the DDS on Pounder and use that for frequency/phase hops (ignoring the RF output from pounder)? (If not, ignore the below)

What size of frequency/phase steps and settling times (loop bandwidth) are you after? If you want large steps, might be worth considering a PFD instead of the mixer used on pounder (or, arguably, we could just put a PFD on Pounder instead of the mixer).

Out of curiosity, do you have e.g. an AOM in your beam path that you could use for the frequency/phase tuning instead? That would give you basically instant tuning rather than having to wait for an analog loop to settle.

hartytp, 2020-01-02:
Anyway, that aside, there is a lot of common ground between what you want to achieve and what we do. For the ~6GHz stuff I suspect we could cover both use-cases on a single board with a few 0402 solder jumpers and population options. 9GHz is getting tough though and would likely warrant a separate design I suspect.

airwoodix, 2020-01-02:
@hartytp Thanks for the feedback and Happy New Year!

> I see. So, the idea is to beat the IF from your Mirny AFE against the DDS on Pounder and use that for frequency/phase hops (ignoring the RF output from pounder)?

Yes, thereby fully reusing the Pounder development and only tackling the high frequency part here. It also now sounds more reuseable to me to just have a heterodyning AFE than something with 1000 options. Is it reasonable for you to use a Mirny AFE to downconvert the frequency to a band that can be handled by Pounder and re-use it completely? I'd happily see this as a generic method to do MHz-GHz frequency offset/phase locks, unless there's a drawback I didn't consider. I'm mostly worried about the necessary firmware developments for Stabilizer but if this is anyway done for Pounder it's fine (and easier to share the load).

A variant of Pounder with a PFD (e.g. ADF4154) would be cool but probably unnecessary if Pounder exists. Nevertheless, would you see this as a feature/hardware option for Homodyner?

We don't have any application above 7 GHz. I only mentioned the 9.2 GHz for neutral Cs as it might increase the number of potential users. If it turns out to be (too) annoying for engineering reasons, I'd drop it.

(We typically rather want to sweep the phase than jump it (e.g. to compensate for the changing Doppler shift as the atoms fall down in the case of a gravimeter). On the current setups, the beatnote is typically mixed with a 7 GHz reference and this signal is then phase locked to a DDS with a homebrew board (https://bit.ly/36m3Pq8). On the large scale experiments, you need to use two lasers as the range of an AOM is not sufficient.)

hartytp, 2020-01-02:
> Is it reasonable for you to use a Mirny AFE to downconvert the frequency to a band that can be handled by Pounder and re-use it completely?

It's an option, but less convenient -- I'd rather just have Stabilizer and a single AFE than: Stabilizer+Pounder+Mirny+AFE as well as the wiring between them and something to drive Mirny (Kasli? Humpback? Stabilizer EEM connector and write Stabilizer firmware to control Mirny?).

> I'm mostly worried about the necessary firmware developments for Stabilizer but if this is anyway done for Pounder it's fine (and easier to share the load).

Part of the motivation for our approach is that it requires very little firmware development. We can largely reuse the Pounder firmware, just replace the DDS part with PLL part (which is basically a python->rust port of the existing artiq driver). AFAICT, this is less work than any other approach, but I might be wrong?

> A variant of Pounder with a PFD (e.g. ADF4154) would be cool but probably unnecessary if Pounder exists. Nevertheless, would you see this as a feature/hardware option for Homodyner?

I think that either/both of Pounder/Homodyner could easily have a PFD population option if someone articulates a use-case for it (just open an issue on the relevant repository).

> We don't have any application above 7 GHz. I only mentioned the 9.2 GHz for neutral Cs as it might increase the number of potential users. If it turns out to be (too) annoying for engineering reasons, I'd drop it.

AFAICT things rapidly get complex above ~6GHz and many of the components one might want become hard to find at 9GHz, so I wouldn't do that unless there is an actual need -- and besides which, there are always more applications at higher frequencies, so one has to stop somewhere. 

> (We typically rather want to sweep the phase than jump it (e.g. to compensate for the changing Doppler shift as the atoms fall down in the case of a gravimeter). On the current setups, the beatnote [...]

Ok, makes sense. For us, I don't think this kind of approach is particularly useful since the phase/frequency sweeps have to be quite slow to allow the PLL to track them accurately. When we want things frequency agile, we typically want 1us jumps, which is too fast for a PLL (hence using a separate modulator driven by a DDS). So, for us, I don't see any benefit of a two-stage mix-down process, which is why it makes sense to put everything on the same AFE.

--- 

Anyway, this has been a useful discussion and I now think I understand what you're up to, so thanks for that.

It's looking like we're going to make a "heterodyner" as a Stabilizer AFE (with minimal modifications to Pounder) at some point after Pounder's been tested. If that sounds useful to you then let's join forces. The other approaches you outline are obviously fine too...

hartytp, 2020-01-02:
> It's an option, but less convenient -- I'd rather just have Stabilizer and a single AFE than: Stabilizer+Pounder+Mirny+AFE as well as the wiring between them and something to drive Mirny (Kasli? Humpback? Stabilizer EEM connector and write Stabilizer firmware to control Mirny?).

A bit of extra context: we're looking for a fixed frequency offset lock between a pair of lasers. So, it's useful to have something that can easily run standalone (rather than plugging into one of our ARTIQ Kaslis). Hence a PoE module like Stabilizer is ideal for this (although you could still connect it to a Kasli, albeit with limited timing resolution, using the Stabilizer EEM).

gkasprow, 2020-09-30:
@hartytp what are your needs here? Shall we proceed? The Almazny is nearly done and we can take care of this design.

gkasprow, 2020-09-30:
@airwoodix

hartytp, 2020-09-30:
@gkasprow for our specific application, the best method would still probably be a specific Stabilizer mezzanine. As described above, this would be something like:
- Two PLL ICs
- One clocked from an external reference, used to feed mixer LO
- the other clocked from a VCXO that is tuned by the stabilizer DAC to lock
- RF input/mixer/signal conditioning largely copied from Pounder, but changing the bandwidth of some components to work to a few GHz

The current Mirny design gives us a way of building something like the above using a stabilizer + mirny + containerized mini circuits bits. Modulo concerns about the cross-talk from the clock jumpers and CMRR of the VCXO input, I think this does everything we could want without building a dedicated card (which may be more effort than it's worth).

airwoodix, 2020-09-30:
@gkasprow are the schematics of the Almazny already online?

gkasprow, 2020-09-30:
@shanasz has issues with github, they should arrive today.

closed 2020-09-30 by gkasprow