sinara-hw/mirny #23: More flexible clock routing

hartytp, 2020-02-24:
https://github.com/sinara-hw/mirny/issues/10#issuecomment-558618102

gkasprow, 2020-09-30:
@hartytp @jordens would such wiring work for you?
[image]

hartytp, 2020-09-30:
That arrangement allows one to select between the SMA and the VCXO/MMCX on a per-channel basis. For our use-cases, it would be more convenient to select between the MMCX and the SMA/VCXO on a per-channel basis, but the above is acceptable. @jordens preferences here?

@gkasprow based on our previous experience, I guess we expect quite a bit of cross-talk from the DNPd jumpers, do we?

gkasprow, 2020-09-30:
we can use T-type jumpers, when unused, the central part is grounded.

jordens, 2020-09-30:
Either is fine by me.

jordens, 2020-09-30:
Default population being all channels clocked from IC13.

gkasprow, 2020-09-30:
I added such a scheme. To avoid dangling nets, the segment is terminated on both sides. To improve isolation, unused net segments can be grounded. That should solve the issue with crosstalk.
[image]

gkasprow, 2020-09-30:
I also changed the assignment - now we select between MMCX and the SMA/VCXO.

gkasprow, 2020-09-30:
The layout/routing of the clocking section
[image]

closed 2020-09-30 by gkasprow