sinara-hw/mirny #24: VCO tune CMRR

hartytp, 2020-02-24:
https://github.com/sinara-hw/Urukul/issues/37#issuecomment-588092910

To do: try driving the VCO from a stabilizer and measure the DDS output noise spectrum. If it's bad, look at putting in a proper InAmp solution with much better CMRR (and maybe a CM/DM EMI filter).

gkasprow, 2020-09-30:
I took the Urukul [issue](https://github.com/sinara-hw/Urukul/issues/37) into account
[image]

hartytp, 2020-09-30:
This would be clearer with the population options shown, but I think I understand what's going on...I'd feel more confident with this if we'd tested on Urukul, but the approach seems sensible (assuming we can get away without a true differential front end to suppress ground noise)

hartytp, 2020-09-30:
@gkasprow what limits the tuning bandwidth here? Is it C100? I assume we want a filter to stop RF pickup etc getting onto the VCO.

gkasprow, 2020-09-30:
default population:
[image]
the issue is with filtering the + divider - for low frequency if will have more than 5k and its CMRR for low frequency would be worse. I'd prefer to stay with one used in Urukul.

hartytp, 2020-09-30:
don't C100 + R78 give an nice filter?

gkasprow, 2020-09-30:
it does. But in one of the Urukul issues somebody wanted additional RC filter at the opamp positive input. That's why I added R99, R100 and C107

hartytp, 2020-09-30:
Fine. Well, we have a pretty sensible topology and lots of pads to play with here, so it seems like a decent starting point. Would be good for someone to actually play with a phase lock...

closed 2020-09-30 by hartytp