sinara-hw/mirny #25: High frequency mezzanine

cjbe, 2020-07-25:
How far advanced are the plans for the high frequency mezzanine?

I need some sources around 5-12 GHz for which Mirny+mezzanine would be ideal. For my use case all I would require on the mezzanine is matching and switching. Switchable attenuation would be a nice to have.

Does anyone have any appetite to specify / use / build this? If so I can put some effort into crystallising specifications.

gkasprow, 2020-07-25:
@cjbe So far there was no such request.
If you specify it, we will build it. I have a budget for RF engineers and prototypes.
Just find suitable RF components and write the spec.

gkasprow, 2020-07-25:
U can use a free RF QUCS simulator to do a rough estimation if the components make sense.
This is an example project of ISM transverter I designed and my colleague verified with this tool
[image]

cjbe, 2020-08-01:
Here is a broad specification:
* max output power: ~5-10 dBm (enough to match well to standard 25-35 dB gain amps)
* frequency range: 4 GHz - 13 GHz (ie the complement of range covered by the main outputs)
* switching with extinction >~ 50 dB

Optional:
* step attenuator with >15 dB range, <1 dB LSB

### Components
All chosen from Analog / HMC

#### RF Switch
The best option looks like [HMC1118](https://www.analog.com/en/products/hmc1118.html) ($7):
Range 0-13 GHz
Typical IL 1.3 dB
Off isolation: 35 dB @ 10 GHz, 50 dB @ 4 GHz
Supplies: +3.3V, -2.5V
All other options are more expensive, give only marginally better extinction, and still require a negative rail

The extinction at 13 GHz is lower than desirable - hitting our target may require using two switches in series.

#### Amplifier
[AVA-183A+](https://www.minicircuits.com/pdfs/AVA-183A+.pdf) ($8):
Range 0.5-18 GHz
gain: 14 dB @ 10 GHz
P1dB: 19 dBm @ 10 GHz
Supply: +5V

#### Attenuator
[HMC424ALP3E](https://www.analog.com/media/en/technical-documentation/data-sheets/hmc424a.pdf) ($15)
Range DC-13 GHz
0-31.5 dB in 0.5 dB steps
6 bit control bus
Supply: -5V
IL: 4 dB @ 10 GHz

Can drive with a 74HC595 shift register, giving an SPI interface that looks very similar to the attenuators in the 0-4 GHz Mirny path.

cjbe, 2020-08-01:
On the v1 Mirny there are only 8 digital IO signals - this is fewer than desirable for a fully capable 4 channel mezzanine.

The RF switches need dedicated IO lines (so total 4x), leaving only 4x lines for the 4 attenuators. To fully and independently connect these requires 5x IO lines each, giving a total of 20 lines. Without loosing any flexibility of use, we could reduce this to 3x lines each (data, clock, latch), requiring a total of 12 lines. This requires an extra 8x IO lines.

It looks like we can find 8 free IO on the CPLD (using 7 NC, and 1 from a TP). The neatest way to do this would just be to use a larger 0.1" header; this would require breaking backwards compatibility, but I assume there are few enough of these boards in the wild that this is not a great concern (thoughts @gkasprow @jordens ?)

gkasprow, 2020-08-12:
Yes, we plan new revision. No problem adding other IOs. I opened dedicated issue.  There were so far only 4 or 5 boards produced.
My coleague (Staszek) started working on this mezzanine project.

gkasprow, 2020-08-12:
@cjbe Any idea for the mezzanine name? I'd like to open dedicated repo.

sotirova, 2020-08-13:
How about Almazny? It sounds somewhat similar to mezzanine and is close to Mirny...
[image]

jonathanpritchard, 2020-09-02:
We would also be interested in a mezzanine with this specification, also agree two switches would be better to achieve sufficient high frequency isolation

cjbe, 2020-09-02:
@jonathanpritchard great!
@gkasprow what do you need from us to get this moving?

gkasprow, 2020-09-14:
I assume this would be 4-channel design.
I think we have all is needed to start design process.

gkasprow, 2020-09-14:
@shanasz is our microwave expert, he takes care of the RF design.