sinara-hw/mirny #36: Output crosstalk due to lacking isolation

nkrackow, 2022-12-20:
The outputs of current Mirnies V1.1 suffer from significant crosstalk:
[image]

Pushing on the PCB close to the front panel makes the crosstalk vary by more than 10 dB. 
The SMA shield is shorted to the front panel at DC. 

This implies that there is a problem with the front panel, probably the SMA pins making connection.
The same issue appears on other high frequency sinara modules: https://github.com/sinara-hw/Urukul/issues/68

@gkasprow @akaminska

kaolpr, 2022-12-20:
I believe SMA shield is just shorted to FP in a defined assembly variants:
[image]
[image]

gkasprow, 2022-12-20:
yes, but the impedance of short-circuit may be non-negligible for 1GHz and above

akaminska, 2022-12-21:
We will try to re-create and investigate the problem

gkasprow, 2022-12-21:
To do this, we need more info about the setup. Preferably some photos depict how cables are placed.

jordens, 2022-12-21:
@nkrackow sent a photo to @akaminska AFAIK

gkasprow, 2023-03-22:
we plan new revision. What if we remove the assembly variant and connect the SMA connectors to solid GND?

gkasprow, 2023-11-03:
added direct connection like in Urukul

closed 2023-11-03 by gkasprow