sinara-hw/mirny #41: Phase noise measurements

maria-nowicka, 2024-07-27:
**The phase noise of the microwave synthesizer was measured using the internal oscillator in Mirny, and then compared when the module was clocked by the Kasli controller. 

You can find the phase noise measurement for different frequencies in this table:**

[image]

**For the same configurations as above, the table below shows: peak to peak, mean, and standard deviation.**

[image]

**The graphs below are an extension of the tables.**
[image]

[image]

[image]

[image]

[image]

**Here is the setup used for the measurements:**
[image]

gkasprow, 2024-07-28:
It looks like we have at least 3 issues here: DC/DC converter noise on Mirny, DC/DC converter noise on Almazny and Kasli DC/DC converter noise. I can see the first and the third harmonics.

gkasprow, 2024-08-14:
I verified Mirny in the same setup (same controller, same power unit, same cables) but with E5052B SSA and in my home lab (no RF sources nearby)
And it looks completely different.
[image]

gkasprow, 2024-08-14:
Mirny driven from Kasli
[image]

gkasprow, 2024-08-14:
I managed to recreate same plots.
One needs to switch units of spurious from Normalised to Power
[image]

gkasprow, 2024-08-15:
I was playing with LC filters after the 5V5 converter and it looks like the 1.3MHz noise comes through the ground loops.
We have two more DC/DC covnerters but they are within the ADI module and we can't see them in the phase noise spectrum
The 3kHz noise is also caused by the 1.3Mhz converter - it produces ripples of such frequency.
I'll dig deeper into it because we have such supply in many other designs.