sinara-hw/mirny #42: new DIOT version

gkasprow, 2024-08-31:
A new DIOT/EEM version is on the way:
- take the existing design and replace CPLD with ICE40 - circuit and EEM connectivity copied from Fastino-DIOT (with proper power sequencing)
- get rid of LVDS translators
- move power supplies away from DDS chips, we have a lot of board area
- make sure all points are checked: https://github.com/sinara-hw/DIOT-Template/issues/1

gkasprow, 2024-11-04:
@micbeldyga

The design looks good. Issues:

- name it Mirny DIOT v 1.6
- add legend to SPI connector and DIP switches
- add DIOT panel (6HP) as DIOT assembly option; 
- add SPM connectors legend
- add EEM and DIOT variant, add adf5356-DIOT variant, existing ones keep as EEM variants
- make sure to select IC22 or IC24 in variants

micbeldyga, 2024-11-09:
Done 6ea116b25847b98eab5f359c992dccd5cc83ebca